moode-player/moode #682: Thumbnails not displayed for large library

jcor29, 2024-09-13:
Hi
Thumbnails are not displayed properly for large audio lib. (6T)
When you first add a repo with a few thousands of track this works fine
But if you load libraries with over 100000 tracks, thumbail are not rendered properly
This is happens in the web UI
[image]
The sames happens with the tag view
Note: I tried to regenerate thumbnails , this is the same

May be a command or script could be handy to do so (just in case)

moodeaudio, 2024-09-13:
That screen shot suggests that the thumbnail images were not generated at all. Maybe the thumbnail generator crashed for some reason.

You could try looking in the moode log and MPD log for errors.

jcor29, 2024-09-13:
Ok, I'll check it out.

closed 2024-10-28 by github-actions[bot]