moode-player/moode #694: Allow specifying librespot zeroconf port

curiousCephalopod, 2024-12-05:
I run mo0de on a separate VLAN to client devices. The port 5353 broadcast is easily handled by udp-broadcast-relay but the zerconf port is randomly generated on start of librespot which makes configuring the firewall a major pain.

Librespot allows this port to be fixed with a command line flag `--zeroconf-port <int>`:
https://github.com/librespot-org/librespot/issues/91

Requesting an option to specify this in the mo0de Spotify Connect renderer settings.

moodeaudio, 2024-12-05:
If there are a significant number of requests for this I can add it to my chronically overcomitted TODO list but otherwise I'll have to leave it up to another dev to implement.

There are a number of source files and resources that need to be updated in order to add a new option for librespot in case someone wants to take it on.

1. cfg_spotify SQL table
2. spo-config.html
3. spo-config.php
4. renderer.php (lib containing startSpotify() function)
5. autocfg.php (backup/restore)
6. sysinfo.sh (system info report)

moodeaudio, 2024-12-18:
PR merged and tested.

closed 2024-12-18 by moodeaudio