moode-player/moode #696: Support for ultra-wide local displays 11.9" 1480x320

Stephanowicz, 2024-12-07:
Hi!

Could You please add media support for those ultra-wide screens (1480x320) like this:  
[https://www.waveshare.com/wiki/11.9inch_HDMI_LCD](https://www.waveshare.com/wiki/11.9inch_HDMI_LCD)  

The problem is that in the main-screen the buttons get cut off:  
[image]  
  
I made a fix so it would look like this:  
[image]  
   
it shouldn't be too much effort to add this to the css.

Thank You & cheers,

Stephan

Stephanowicz, 2024-12-07:
FYI: this is the fix I added ( as I didn't look to deep into code and didn't find where the canvas for the countdown is created I overruled it with '!important'.)  

Just a memory aid for where to start :)

```
				@media screen and (max-width:1480px) and (max-height:320px) and (orientation:landscape) {
					#countdown > div {
						height: 180px !important;
						width: 180px !important;
					}
					#countdown > div > canvas {
						height: 180px !important;
						width: 180px !important;
					}
					#playback-controls > #timezone {
						height: 44%;
					}
					#timeknob > #countdown-display {
						position:absolute;
						top:34%;
						left:50%;
						margin-right:0px;
						margin-bottom:0px;
						transform:translate(-50%, -50%);
						font-size:2.25vw;
						font-weight:500;
						cursor:pointer;
					}
					#timeknob > #total {
						position:absolute;
						top:45%;
						left:50%;
						transform:translate(-50%);
						font-size:1.5vw;
					}
					#playback-controls > #playbtns {
						display: block;
						height: 17%;
					}	
					#playbtns .btn.btn-cmd {
						font-size: 4.0em;
					}
					#togglebtns .btn-group {
					  font-size: 1.7em;
					  white-space: normal;
					  width: 100%;
					}
				}

```

psy501, 2024-12-07:
Thank you very much, it helps me a lot with my Moode audio DAC project. I've just tested your modification, and I think it's still a bit missing :-(
[image]
[image]

psy501, 2024-12-07:
In which file did you make this modification ?

moodeaudio, 2024-12-07:
My TODO list is chronically overcommitted and so some other dev is going to have to do the work and submit a PR but be sure follow the same coding as in the `// Ultra-wide screens` section including for `/* Time knob */` so itI can be maintained  going forward otherwise it won't be accepted.

Stephanowicz, 2024-12-07:
Please stay in this thread regarding the addon stuff! I don't want to "spam" the moode thread :)

ok - the settings are in script.js starting with line 87 --> addosCfg['uwStyle'] ...   

so I would try following changes:
-------------------------------------------------------
This changes the size of the circle:  
```

#countdown > div {
	height: 180px !important;  --> change it to 176px
	width: 180px !important;  -->change it to 176px
}
#countdown > div > canvas {
	height: 180px !important;  --> change it to 176px
	width: 180px !important;  --> change it to 176px
}
```
-----------------------------------------------------
this changes the position of elapsed and total time in the circle:

```
#timeknob > #countdown-display {
	position:absolute;
	top:34%;	--> change it to 31%

#timeknob > #total {
	position:absolute;
	top:45%;  --> change it to 43%
```
------------------------------------------------------
This changes the position of the play controls:   

```
#playback-controls > #timezone {
	height: 44%; -->change it to 41%
}
```
--------------------------------------------------
This changes the position of the buttons:   

```
#playback-controls > #playbtns {
	display: block;
	height: 17%;  --> change it to 14%
}	

```

Stephanowicz, 2024-12-07:
Sorry - why is this now linked???

Stephanowicz, 2024-12-07:
Ok, I follwed the media.css - there's not much that needs to be adjusted:
```

/*
// Ultra-wide screens 11.9" 1480x320
*/
@media screen and (max-width:1480px) and (max-height:320px) and (orientation:landscape) {
	/* Time knob */
	#countdown {height: 11vw;width: 11vw;}
	#countdown-display {top: 44%;font-size: 2vw; }
	#total {top: 62%;font-size: 1.25vw;}
	#timeknob {margin: 0;}

	/* Play and toggle buttons under the time knob */
	#playbtns .btn {font-size: 2vw;padding: 1rem 2rem;}
	#togglebtns .btn-group {font-size: 1.7em;width: 75%;}
}
```
[image]

Hopefully someone finds the time take care of it :)

Cheers,
Stephan

monkeypox, 2024-12-07:
I forget how the html is set up but on ultra wide screens like that I wonder if it’d be better to have the toggle buttons on the side of the dial control? You could maybe force its position and change the h/w and it should just flow, maybe? There’s an aspect ratio media query that’d make this more general purpose, ***@***.***/aspect-ratioAlso I know the project is kind of constrained by using third party libraries but ideally all css should live in the css files, it’d be cool if you or someone could move that over from the javascript.If this is of general need for ultra wide displays it should probably go in the main branch rather than being an addon. You can use something like vh & vw rather than hard coded pixel values so it automatically adjusts to the screen size, see: https://www.sitepoint.com/css-viewport-units-quick-start/Sent from my iPhone

Stephanowicz, 2024-12-07:
> I forget how the html is set up but on ultra wide screens like that I wonder if it’d be better to have the toggle buttons on the side of the dial control? You could maybe force its position and [...]

Hi, 
maybe You haven't seen my response before - I posted the necessary css that needs to be added to media.css  

Unfortunately my thread seems to be linked here...? there the discussion is about adding addons without changing the moOde source - so that's why it's in js

Sorry for the confusion :) 

Cheers, Stephan

psy501, 2024-12-08:
> Please stay in this thread regarding the addon stuff! I don't want to "spam" the moode thread :)
>
> ok - the settings are in script.js starting with line 87 --> addosCfg['uwStyle'] ...
>
> so I would [...]

Thanks 👌 this IS perfect for my screen [image]

closed 2025-01-23 by github-actions[bot]