moode-player/moode #697: Question:  shell_execs

AlfredoCubitos, 2024-12-08:
Hi, 
I found so many `shell_execs` in your code. 
Is there a special reason why you use `shell_exec` instead using internal php functions?
I think in some cases it would be faster to use native functions instead.

moodeaudio, 2024-12-08:
No special reason, prolly just continuing with whatever the original sources had 12 years ago. 

Scanning the source tree it looks like 39 occurrences of "shell_exec" in 7 files. 

1. Which ones do you think should be replaced?
2. What is the exact code?

[image]

AlfredoCubitos, 2024-12-08:
ah, ok. Thank you!
Do you have plans to switch from php to any other programming language?
I'm asking if  its still worthy to spend some lines of php code ;-)

moodeaudio, 2024-12-08:
You mean rewrite the PHP codebase in some other programming language and convert to other framework libs. I don't see what could possibly justify such a massive and incredibly risky undertaking.

A quick count shows 61176 lines PHP and 5626 lines HTML for total __LOC = 66802__
```
pi@moode9:~ $ cd /var/www/
pi@moode9:/var/www $ find . -name '*.php' | xargs wc -l
     62 ./inp-config.php
    312 ./net-config.php
.
.
.
    338 ./command/queue.php
    165 ./command/cfg-table.php
    142 ./command/index.php
  61176 total
pi@moode9:/var/www $ 

pi@moode9:/var/www/templates $ find . -name '*.html' | xargs wc -l
   339 ./mpd-config.html
   495 ./snd-config.html
.
.
.
   190 ./lib-nas-config.html
    21 ./sys-status.html
  5626 total
```

AlfredoCubitos, 2024-12-08:
I'm just asking to be sure.

Noo, a complete rewrite is too much, and not necessary.
But a few small improvements wouldn't hurt ;-)
Especially the thumb_nail generation and the samba connection need some work. On my Raspi 4 its very slow :-(
But I would appreciate if there is a different way to add changes  as by submitting a pull request ;-)

Btw, I found two possible issues in `thumb-gen.php`:
In line 77 and 78 you should use `ls -d ` to get only directories. 
This is by the way one reason to use php native functions :-)

moodeaudio, 2024-12-08:
I'm always interested in improvements but its best to submit a tested PR and then I'll have a look at the code changes.

closed 2024-12-09 by AlfredoCubitos