moode-player/moode #705: Pleezer login fails

redm123, 2025-03-23:
Hi! I wanted to try out the Deezer Connect feature. However, I enter email and password, start the service, but it doesn't work (a message in the UI would be nice...).  In `/var/log/moode_pleezer.log` I see:

`ERROR pleezer] permission denied: email or password incorrect`

When I check the referenced `/etc/deezer/deezer.toml` everything looks as expected. And I verified with the App that the information is still correct.

Random information, that may be relevant: the email address contains a `+` (like foo+bar@example.com), which is sometimes a problem. But as I said, in the Deezer App and web this works fine. 

Moode version: 9.3.0 2025-03-21
Log `moode_pleezer.log`:

```
[2025-03-23T22:29:05Z DEBUG pleezer] Command Args {
        secrets: "/etc/deezer/deezer.toml",
        name: Some(
            "Moode Deezer",
        ),
        device_type: Web,
        device: Some(
            "ALSA|_audioout||S32",
        ),
        normalize_volume: false,
        initial_volume: Some(
            10,
        ),
        no_interruptions: false,
        bind: "0.0.0.0",
        hook: Some(
            "/var/local/www/commandw/deezevent.sh",
        ),
        quiet: false,
        verbose: 1,
        eavesdrop: false,
    }
[2025-03-23T22:29:05Z INFO  pleezer] starting pleezer/0.13.0.e7633a4 (2025-03-16); release
[2025-03-23T22:29:05Z INFO  pleezer] parsing secrets from /etc/deezer/deezer.toml
[2025-03-23T22:29:05Z DEBUG pleezer::player] no bf_secret specified, fetching one from the web player
[2025-03-23T22:29:07Z INFO  pleezer::remote] logging in with email and password
[2025-03-23T22:29:07Z ERROR pleezer] permission denied: email or password incorrect
```

Log `moode.log`

```
20250323 232905 worker: Job deezersvc
20250323 232905 DEBUG: startDeezer(): (pleezer --name "Moode Deezer" --device-type "web" --device "ALSA|_audioout||S32" --initial-volume "10" --secrets "/etc/deezer/deezer.toml" --hook /var/local/www/commandw/deezevent.sh -v > /var/log/moode_pleezer.log 2>&1 &)
20250323 232905 DEBUG: waitWorker(): Wait 2 for dez_config

```

moodeaudio, 2025-03-23:
Questions and troubleshooting for a release are handled in the Support section of our Forum so that all our users can benefit from the discussion.

Visit the Support page at moodeaudio.org for information on how to easily request a Forum account.

closed 2025-03-23 by moodeaudio

redm123, 2025-03-24:
Sorry, I'm happy to help debugging this, but I generally don't like creating forum accounts all over the internet (and you make even that complicated). And then just to report a problem. That's why we have a central Github with a bug tracker...

redm123, 2025-03-26:
Ah, didn't know there is a separate repo. Will do 🫡