moode-player/moode #706: mpd 0.24 high usage

atekin, 2025-03-24:
I observe some high cpu usage with MPD local disk playback. It seems that the issue has been discussed https://github.com/MusicPlayerDaemon/MPD/issues/2241

Is it possible to compile MPD without io_uring 

Please post issues, questions or suggestions here that are related to the development of moOde audio player.
General troubleshooting and support for moOde audio player releases is provided in the moOde Forum. Visit the Support section at moodeaudio.org for information on joining the Forum.

MaxKellermann, 2025-03-24:
Your description is incorrect. There is no high CPU usage. It's just a bogus load value (I/O pressure), but it's not actually using any CPU for that. It's a Linux kernel bug. Sure you can disable io_uring, but it will not reduce CPU usage. It will only avoid the bug that shows high I/O pressure (but there really isn't any I/O pressure because the displayed value is bogus).

atekin, 2025-03-24:
I was only stating what htop usage. I will post the screenshots. I don't think htop or any other utility show values incorrectly, they just read values from the system and display it.

MaxKellermann, 2025-03-24:
It seems my explanation was not good enough for you to understand, but it was worth a try. Good luck, I'm out of here.

closed 2025-03-24 by moodeaudio

dheijl, 2025-03-24:
> I was only stating what htop usage. 

In this case the problem is caused by an unfortunate choice made by one of the io_uring kernel developers that causes a process that is idle waiting for IO to complete to be reported as busy in io_wait. You can read all about it [here](https://lwn.net/Articles/989272/) or on the kernel mailing list where discussions about how to fix the confusion are ongoing.

So one could say that both you and @MaxKellermann are right here.

atekin, 2025-03-24:
@dheijl thanks got it